nexdome/Firmware #3: TA.NexDome.Shutter.ino sketch to big

gintasd, 2019-09-24:
Firmware/TA.NexDome.Shutter/TA.NexDome.Shutter.ino sketch shows to big.
[image]

NameOfTheDragon, 2019-10-07:
You can't build this sketch in Arduino IDE. Please see the ReadMe file for information about the required build environment and a discussion of why it was done that way. Also, the code size IS a very tight fit and it might be sensitive to the version of the compiler that you have installed.

gintasd, 2019-10-08:
Hello Tim,
Thank you for answer.
I am not a programmer and have very little experience with the
VisualStudio. Probably I could learn it but it would take to long :).
I will not bother you but just now I stuck with a "Processing -File C*:
_Programos\Astronomija\Observatorija failed because the file does not have
a .ps1 extension. Specify a valid PowerShell script file name, and then try
again" and can not find where to set this.
 Basically I can upload the hex file from the github but I need to change
the gear ratio to align the software to my own dome HW.
Maybe I am doing something wrong but I can not upload the hex file using
the "NexDomeControlSystem.3.0.1".
I have the Arduino port selected but after "Update""button the Bootloader
com port does not open despite the application reboots the Arduino.
And nothing happens.
I did upload successfully using avrdude. After upload the control basically
works fine.
My PC is Win7 64bit. Arduino IDE 1.8.5

Best Regards
Gintautas

NameOfTheDragon, 2019-10-08:
You shouldn't need to recompile for that. The firmware doesn't have any concept of a gear ratio. It just knows the total number of whole steps per dome revolution. Can you not just change the number of steps per revolution using the `@RWR` ("range write") command? Once you've determined the correct number of steps you can save it to EEPROM using `@ZWR`.

gintasd, 2019-10-09:
Aaa, it works this way. OK, clear.
So should I use Full Steps on the motor driver as well if  the range in the
controller software is set by Full Steps?
Or the Full Steps are recalculated in to the Microsteps (x8) what usually
are used?

NameOfTheDragon, 2019-10-14:
I'm not totally clear what you are asking here, but in general as an end user, you always talk to the firmware in whole steps. Microstepping is fixed internally at 8 microsteps per step but you never get to see that as the user. We decided to fix it at 8 uSteps/Step as we didn't want to deal with the support issues that variable microstepping would generate ;-) Sometimes more options just confuses the users.

gintasd, 2019-10-14:
Hi Tim,
That was what I was asking :). I have tested it already a few days ago. And
I agree with you - more options = more problems :).Thanks a lot.

It is not the end of the world but still unable to upload the firmware
using the "NexDomeControlSystem".
I was wrong in my first email regarding the Arduino reboot after the
"Update" button. Actually nothing happens after the "Update" is pushed.
I can see in the window appears the printed string. But I see it tries to
send the update using the same port ( the option -pcom) at what Arduino is
connected.
I mean it does not reboot the Arduino to open the bootloader port.
It updates just fine using the command line.

Best Regards
Gintautas Drazdauskas

NameOfTheDragon, 2019-11-29:
Closing issue as there's no action resulting from it.

closed 2019-11-29 by NameOfTheDragon