nexdome/Firmware #4: Events question

knro, 2019-10-07:
Thank you for the great documentation. I'm developing the INDI driver for NexDome v3 and the documentation is pretty detailed and straight forward.

Regarding the events, what's the delimiter? Is it <CR><LF>

And just to confirm, the XBee, Rotation, and Shutter events are not terminated by a '#'?

NameOfTheDragon, 2019-10-07:
This is related to #2 . I'll be looking into that this week. The intended principle is that all responses will be encapsulated between `:` and `#` (but see #2).

Where an end-of-line is used, the firmware accepts <CR> 0x0D or <LF> 0x0A or both, in any order. Clients should probably follow suit.

knro, 2019-10-07:
If everything is between : and # including all events then it would be great and consistent with the rest of the command protocol. I'll develop the driver based on the latest firmware (3.0?) and will update it once a firmware update is released.

NameOfTheDragon, 2019-10-08:
There is a new firmware build for you to try in PR #5

NameOfTheDragon, 2019-10-14:
For the time being at least (mainly due to code size considerations) the firmware feature set is frozen.

closed 2019-10-14 by NameOfTheDragon