nexdome/Firmware #8: AtHome flag may be false when FindHome completes

NameOfTheDragon, 2019-10-22:
This problem was raised when a user observed the dome apparently trying multiple times to find the home sensor. The user was using the X2 driver.

Rodolphe reports:

> It retries because when it stops it’s not on the home magnet, it backs too much depending on speed and acceleration Even without myb plugin by using a simple serial terminal I can reproduce the home miss, When I get back home tonight I’ll make a small movie of what happens when the “dome” backs back to the home sensor and the speed is too high and/or acceleration is too long.
This is 100% reproducible and when you send a @RRR the atHome flag is 0.

To which I have replied: 

> OK so I think you’re checking the home flag at the end of all that and seeing it as false. That might be a bug, I’ll check into it. During the development I changed how homing worked because my original method didn’t work very well. The method I finally settled on works very well and has good repeatability, but It’s possible that I’m not handling the home flag correctly.

rpineau, 2019-10-22:
From you email :  "When the dome stops, it is _by definition_ at the home position.”
then the home flag should be reporting At Home in the :SER,p,a,c,h,d# answer :

a = AtHome (1 = home sensor active, 0 = home sensor inactive)

May be this shouldn’t report the sensor active or not but At Home state which from an application stand point is what we care about when checking if the dome is “at home”.
Or if this is working as intended (actually reporting sensor state), then we need a proper command to ask the controller if the dome is At Home .
May be in ASCOM you assume that if the dome has stopped moving it’s at home, but I thing we should be able to asses this via a command or a state so that if the user click on “Find Home” we can avoid sending the @GHR command if the dome is already at home and make sure that when it stops it's because it did find home and not stop for other reason somewhere than is not home. So we need a way to check that the dome is indeed at home.
In the mean time I’ll remove the retry from my code to avoid sending the dome around multiple times for no reason.

dave4robots, 2019-10-22:
I verified the magnets are in the proper place. I opened the SkyX dome controller and sent it to various azimuths and told it to home. It homed each time without incident. I could not replicate the problem.

This AM the problem of continuous clockwise happen with every home attempt and I tried about 5 times. I eventually gave up on home and just sent it to Park, which was several degrees off. I manually eyeballed home position because I was working from home and monitoring on my web camera. I got it to stop at the charge points and powered down.

This afternoon after a restart it appears to go to home position okay. So I wish I could provide more data but unable to replicate. 

Best of luck and thanks for all the support so far both Tim and Rodolphe.

rpineau, 2019-10-22:
> 
> I verified the magnets are in the proper place. I opened the SkyX dome controller and sent it to various azimuths and told it to home. It homed each time without incident. I could not replicate the problem.
Weird.

> This AM the problem of continuous clockwise happen with every home attempt and I tried about 5 times. I eventually gave up on home and just sent it to Park, which was several degrees off. I manually [...]

Could the fact that it end up being off by several degree also impact the homing ?
In any case if you want to avoid the retry that send the dome retating over and over, use the new version where this is disable.
It’ll do one homing and if it’s not on the home magnet when the dome stop it will report an error instead.

> This afternoon after a restart it appears to go to home position okay. So I wish I could provide more data but unable to replicate.
I’ll run some test tonight and will do a lot of homing at different speed.

> Best of luck and thanks for all the support so far both Tim and Rodolphe.

No problem.

Rodolphe

rpineau, 2019-10-23:
Here is the log from my plugin that shows that after backing it doesn't see the home sensor every time : 
<pre>
[Tue Oct 22 17:32:02 2019] [CNexDomeV3::isFindHomeComplete]
[Tue Oct 22 17:32:02 2019] [CNexDomeV3::isDomeMoving] In : m_bDomeIsMoving = Yes
[Tue Oct 22 17:32:02 2019] [CNexDomeV3::isDomeMoving] szResp = P11987
[Tue Oct 22 17:32:02 2019] [CNexDomeV3::isDomeMoving] szResp = P12286
[Tue Oct 22 17:32:02 2019] [CNexDomeV3::isDomeMoving] szResp = foundHome
[Tue Oct 22 17:32:02 2019] [CNexDomeV3::isDomeMoving] szResp = P145
[Tue Oct 22 17:32:02 2019] [CNexDomeV3::isDomeMoving] szResp = P368
[Tue Oct 22 17:32:02 2019] [CNexDomeV3::isDomeMoving] Out: m_bDomeIsMoving = Yes
[Tue Oct 22 17:32:02 2019] [CNexDomeV3::isFindHomeComplete] still moving
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isFindHomeComplete]
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isDomeMoving] In : m_bDomeIsMoving = Yes
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isDomeMoving] szResp = P517
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isDomeMoving] szResp = P591
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isDomeMoving] szResp = :SER,600,0,12347,0,30#
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isDomeMoving] szResp = onMotorStopped
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isDomeMoving] szResp = Post slew from 4800 to 0
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isDomeMoving] szResp = P580
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isDomeMoving] szResp = P488
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isDomeMoving] Out: m_bDomeIsMoving = No
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::domeCommand] sending : @SRR
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::domeCommand] response : ':SER,420,0,12347,0,30'
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isDomeAtHome] response = SER,420,0,12347,0,30
[Tue Oct 22 17:32:03 2019] CNexDomeV3::isDomeAtHome bAthome : False
[Tue Oct 22 17:32:03 2019] [CNexDomeV3::isFindHomeComplete] Not moving and not at home !!!
[Tue Oct 22 17:32:04 2019] [CNexDomeV3::domeCommand] sending : @PRR
[Tue Oct 22 17:32:04 2019] [CNexDomeV3::domeCommand] response : 'P321'
</pre>
To get this result I set the speed to 1200 step/s and the acceleration to 1000ms.

If I use my previous value of 600 step.s and 500 ms acceleration here is the log :
<pre>
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isFindHomeComplete]
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] In : m_bDomeIsMoving = Yes
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] szResp = P16
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] szResp = :BV821#
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] szResp = foundHome
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] szResp = P67
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] szResp = P141
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] szResp = :SER,150,0,12347,0,30#
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] szResp = onMotorStopped
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] szResp = Post slew from 1200 to 0
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] szResp = P130
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeMoving] Out: m_bDomeIsMoving = No
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::domeCommand] sending : @SRR
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::domeCommand] response : ':SER,110,1,12347,0,30'
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isDomeAtHome] response = SER,110,1,12347,0,30
[Tue Oct 22 17:37:31 2019] CNexDomeV3::isDomeAtHome bAthome : True
[Tue Oct 22 17:37:31 2019] [CNexDomeV3::isFindHomeComplete] At Home
 </pre>

With the slower speed and lower acceleration (and deceleration time) the magnet is ALWAYS detected.

I can repeat the 2 above test over and over and always get the same result.
In one case the At Home flag is 0 (faster speed and longer acceleration), in the other case it's 1 (slower speed and faster acceleration).
The end position between the 2 test is very minimal on my rig as it's small but it's enough to show a detected magnet vs not detected and therefore show it as not being at home.

closed 2019-11-30 by NameOfTheDragon