nexdome/Firmware #12: There is no way to perform a hard reset of the XBee

NameOfTheDragon, 2019-10-24:
This is an issue for @nexdome to review and decide whether there is any feasible hardware remedy.

## Problem

The DFRobot Arduino Leonardo with XBee socket leaves the XBee reset pin disconnected. This is OK technically because it has an internal 50K pullup, but it means there's no way for the Arduino to perform a hard reset of the XBee. This is a problem because the XBees sometimes get into a state where they either refuse to associate with another XBee, or refuse to enter command mode so that the firmware is unable to send configuration and the only fix seems to be to power-cycle the system.

As you can see from DFRobot's circuit schematic, the XBee reset pin is left open.
[image]

According to the XBee data sheet:
[image]

The `/RESET` signal is forbidden from being driven high and this is probably why it was left unconnected, since Arduino output pins might drive the signal high after a power-on reset.

However this creates a problem where XBees can sometimes get 'stuck' (despite the firmware issuing a soft reset) and there's no way to recover other than to power off the system, which is quite inconvenient for users.

## Questions

Therefore, the questions I would like to raise for discussion are:

- Would there be any point to connecting up this reset line? Would it in fact let us reset the device?
- Could it be done in such a way that the Arduino can safely activate it?
- If so then, would it be practical to modify the electronics to enable this as part of the production process?

rpineau, 2019-11-15:
On the Teensy proto I made, I did wire the XBee reset for this exact reason, issuing a proper reset to the device. This is equivalent to a power-on reset for the XBee. This could be "fixed" on the current arduino board by soldering a wire under the board between the XBee reset pin and one of the available IO pins (pin #13 is not used and not far from the XBee socket).

NameOfTheDragon, 2020-02-18:
That's interesting @rpineau . Did you have any problems with the output pins potentially being high at power-on-reset of the microprocessor? Or do they default to open drain?

rpineau, 2020-02-18:
When they say never drive it high they mean that as the pin is meant to be shared by multiple devices you need to use a open drain output (aka being able to be set in a tri-state mode, aka High Z).
There is no actual problem with driving the pin high. You can do this by setting the pin on which reset is wired as an input and only setting it LOW and then as an output when you need to assert the reset, then setting it back as an input.
As there is an internal pull-up on the XBee, the pin will be driven high anyway. This is very standard for reset pins and the XBee documentation make it sound like driving the pin high is a bad thing, it's not and will not damage the XBee if that would happen during boot. And then you set it as described above.

NameOfTheDragon, 2020-02-18:
Thanks Rodolphe, that's very clear. I see that the Arduino digital pins default to inputs on power-up, so that would be safe.

I'm not sure if we will need to do this now for the current design but in any future designs I would prefer to have this capability. Then again, I'd prefer not to use XBees ;-)

closed 2020-06-22 by NameOfTheDragon