nexdome/Firmware #18: No response to @PRS commands

rpineau, 2019-11-15:
Hi Tim.
Based on a few lofs for 2 users I see timeout every time I request the shutter position 👍 
<pre>
[Thu Nov 14 20:26:08 2019] [CNexDomeV3::domeCommand] sending : @PRS

[Thu Nov 14 20:26:08 2019] [CNexDomeV3::domeCommand] response : 'XB->Online'
[Thu Nov 14 20:26:08 2019] [CNexDomeV3::domeCommand] XBee status : 'XB->Online'
[Thu Nov 14 20:26:09 2019] CNexDomeV3::readResponse Timeout while waiting for response from controller
[Thu Nov 14 20:26:10 2019] CNexDomeV3::readResponse Timeout while waiting for response from controller
[Thu Nov 14 20:26:12 2019] CNexDomeV3::readResponse Timeout while waiting for response from controller
[Thu Nov 14 20:26:13 2019] CNexDomeV3::readResponse Timeout while waiting for response from controller
[Thu Nov 14 20:26:14 2019] CNexDomeV3::readResponse Timeout while waiting for response from controller
[Thu Nov 14 20:26:16 2019] CNexDomeV3::readResponse Timeout while waiting for response from controller
[Thu Nov 14 20:26:17 2019] CNexDomeV3::readResponse Timeout while waiting for response from controller
[Thu Nov 14 20:26:18 2019] CNexDomeV3::readResponse Timeout while waiting for response from controller
[Thu Nov 14 20:26:18 2019] [CNexDomeV3::domeCommand] response : 'XB->Online'
[Thu Nov 14 20:26:18 2019] [CNexDomeV3::domeCommand] XBee status : 'XB->Online'
[Thu Nov 14 20:26:19 2019] CNexDomeV3::readResponse Timeout while waiting for response from controller
[Thu Nov 14 20:26:20 2019] [CNexDomeV3::getDomeEl] ERROR = PRR20196
</pre>

as you can see I do get some of the inline messages, like the XBee online status and no actual :PRSxxxxx response to the @PRS command, but I do get the same response as the previous command (which was a @PRR in this case).
So there is either a timeout in the XBee communication with the shutter or in the firmware somewhere.
Is there a minimum amount of time required between commands ?

NameOfTheDragon, 2019-11-25:
@rpineau Which firmware version was this please?

rpineau, 2019-11-25:
I believe these log come from people using the 3.0.0 firmware.
I can retest with the 3.1 beta in pull request #17

NameOfTheDragon, 2019-11-25:
I can’t remember what state that pull request is in. There’s a new build due out within a day or two, probably best to wait for that. I’ll email you when I release it.

Best regards,
Tim Long

rpineau, 2019-11-25:
ok, I can wait and retest once 3.1 is out (or a pre-version so that if this is still and issue it can be fixed before 3.1 is released).

NameOfTheDragon, 2019-11-29:
So now that 3.1.0-Beta.6 is out, I'm having another look at this. My eye is drawn to this part of the log:

> [Thu Nov 14 20:26:08 2019] [CNexDomeV3::domeCommand] sending : @PRS
> [Thu Nov 14 20:26:08 2019] [CNexDomeV3::domeCommand] response : ':PRR20196'

You seem to be getting two responses to the `@PRR` command. Do we know where that second response is coming from?

When you send the second one, that's clearly not answered and that is (or was) a bug. I think this one should be fixed by PR #17 . There was a race condition where a command could get lost if it happened right as the XBee state machine was polling for an active connection. This race condition should now be eliminated.

rpineau, 2019-11-29:
I'll re-run some test with log enabled and check if this is still the case.

closed 2019-11-30 by NameOfTheDragon