nexdome/Firmware #19: @GAR to a position that is the current position

rpineau, 2019-11-15:
Not quite sure how to qualify this one.. but, if I'm at position 108 and do a @GAR;108, I get a :GAR response but I would also hope to get a :SER response after to indicate the end of the move even if there was no move and will not get a :left or :right
So it's not a bug per say, more of a behavior question. 
Thanks
Rodolphe

NameOfTheDragon, 2019-11-25:
The status `SRS` message is generated in the `onMotorStop` event handler, so if the motor never starts (e.g. because the target position is within the dead zone) then there will be no stop event and the status message will not be sent.

The way I handle this in the ASCOM driver is that I I don't assume that movement has started just because I requested it. Once I receive a direction event or a position update, then that's the point at which I know movement has started. I know movement has stopped when either:
1. I don't receive any position updates for a period (I think 2 seconds from memory)
2. I receive the status/motor stopped event.

Does that answer the question?

rpineau, 2019-11-25:
yea that's fine I took care of this in my plugin.

closed 2019-11-30 by NameOfTheDragon