nexdome/Firmware #24: slipping sprocket won't close shutter

Rojoyinc, 2019-12-03:
I posted this in the wrong version. Reposting here. (may be it's fixed already) as a update was just released? 

I had an issue where my shutter sprocket slipped a few turns. (I think due to friction build up in the shutter panel). Once the sprocket slipped the shutter position count was in error. When the driver showed and stated the shutter was closed. (it was in reality very much open). There was no way for me to RESET and close the shutter without going out there manually. If this had happened the week before (I was 800 miles away) and it would have been open for days. !!!!

It did it a week later again. I got up at 3am from bed and closed it manually. In doing so (not sure why) pressing the rocker on the shutter box to close would only do so in about 2-3inch increments and then stop. A release and repress of the rocker continued another 2-3 inches and after 20 or so presses it was closed.

NameOfTheDragon, 2019-12-24:
I can't reproduce the "only moving by 2-3 inch increments" issue using the currently released firmware (3.2.0). So I'm going to assume that whatever is causing that is fixed. It would be useful though if you can confirm this after checking that you;ve got the latest firmware build (there will be another firmware hotfix 3.2.1 bundled with the upcoming ASCOM driver update, coming soon).

As for the slipping sprocket, what would you have me do? A slipping sprocket is a mechanical issue. The mechanics have to work or the software doesn't stand a chance. If you would have me create a "software victory over hardware" then that opens a rather large can of worms.

NameOfTheDragon, 2020-02-18:
Our policy is not to try to create "software victory over hardware". Working, correctly adjusted hardware and mechanics is a prerequisite for functioning software. Therefore we cannot address this issue in software.

closed 2020-02-18 by NameOfTheDragon

AstroFrank, 2020-06-12:
Hello Tim, first of I would like to thank you for developing these drivers for our Nexdomes. You are making automation possible and that allows us to travel and have a little bit more sleep. The shutter control is a critical element to prevent safety of the optics's exposure to sunlight and water onto our precious equipment. 

A little more on the issue above. I recently installed the shutter kit and had a few mechanical slips when I first mounted the track railing. I have then adjusted the hardware and the gear is no longer slipping... But the shutter won't close completely anymore. It will not keep going all the way to the Reid switch. The display window shows "open" and yet it's at the zero position. No way to reset anything remotely. On the dome rotation, it is possible to allow the user to manually provide arbitrary adjustments with the goto angle and the sync button/command. This issue with the incorrect shutter position in the software could be alliviated manually by allowing the user to goto and sync the shutter position ? That functionality would not change any current software behaviour but just give more control to the remote user.

NameOfTheDragon, 2020-06-12:
@AstroFrank Thank you for the kind words.

The shutter firmware has some safety features that are designed to stop the shutter "crashing" into the mechanical limits of travel. One of these features is that the shutter will only move to step position 0 before stopping. If you've had some sort of mechanical slippage then step position 0 probably no longer corresponds to the closed position. So you're right that the firmware needs to be resynchronized, 

The firmware is designed to resync itself whenever it sees the close limit sensor activate. If for any reason it is not closing all the way, then use the toggle switch to manually move the shutter all the way closed until it detects the limit sensor and reports "closed" status in the ASCOM driver. At that point, the reported position should be at or very close to zero. Now, if you try opening and closing the shutter, as long as there is no mechanical slip, it should close all the way.

AstroFrank, 2020-06-14:
Hey Tim, I confirm yes this works well. Is it possible to make an interface
button in the driver for the toggle switch to be operated remotely? (I do
use webcams). Another option for me would be to connect an ESP32 on the
toggle switch and use it has a WIFI relay to accomplish same override
behaviour in case of failure.
Thank you