nexdome/Firmware #26: what is current version (3.2.1?) and is there a .hex for it

dave4robots, 2020-01-20:
I am working from 3.2.1 beta and I am still having problems where the shutter gets dropped if the other controller is turned off.

I have been leaving the dome on but if power blinks or anything goofy happens, I have to go up and reset the shutter.

I saw there is a 3.2.1 firmware folder, but I could only find the rotator .ino files. Is there a compiled/ready to go  .hex file for it?

Dave Compton

NameOfTheDragon, 2020-01-27:
3.2.1 hasn't been formally released because of an issue that we are investigating. You may want to watch https://github.com/nexdome/ASCOM/issues/19

NameOfTheDragon, 2020-02-18:
3.3.0 will be released later today, there will be a release with firmware images that you can download. Keep an eye open for it at https://github.com/nexdome/Firmware/releases/latest

There will also be an updated ASCOM driver today or tomorrow that will include this release.

closed 2020-02-18 by NameOfTheDragon