nexdome/Firmware #32: Shutter status after manual activation

w4sm, 2020-08-23:
When the software closes the shutter it appropiately shows "Closed" at the end of the cycle.  However, if you use the manual control to tweak the shutter just a bit further closed, the software reports "Open" even though the magnetic switch is still in the closed position.   Disconnect and reconnect and the software once again shows "Closed."

NameOfTheDragon, 2020-08-24:
With open vs. closed, the fail-safe condition must be to report that the dome is open, since a false closed indication could be disastrous. Therefore, the software uses the following definitions:

1. The shutter is considered `Closed` if and only if the closed sensor is asserted (making electrical connection).
2. `Open` is defined as `not Closed`.

This can sometimes lead to a situation where the shutter visually appears closed by electrically it cannot be verified, so the default fail-safe condition is to report `Open`.

It is possible that manually using the toggle switch to close the shutter beyond where the closed condition has been detected, could lead to it being reported as open.

The way to check this would be to examine the log file to see what the firmware is reporting. Please attach a log file that demonstrates the issue. You can find log files in `C:\users\`your-username`\logs`. Ideally, delete all existing logs, reproduce the problem, record the time that the problem occurred as accurately as possible (so we can correlate it to the log file) and then attach your log file to this issue as a file attachment (do not copy and paste text from the log file).

w4sm, 2020-08-24:
I connected to the dome, started a shutter opening and let it run for a few seconds, stopped the shutter in the software when it was partially open, closed the shutter in the software.  At that point the display properly indicated "closed."  I then manually closed it about half an inch more.  The display showed "open."  I then disconnected.  The magnetic switch should have stayed closed during and after my brief manual change.
[TA.Nexdome.Server-2020-08-24-HEP-PC_Stacey Mills-HEP-PC.log](https://github.com/nexdome/Firmware/files/5118672/TA.Nexdome.Server-2020-08-24-HEP-PC_Stacey.Mills-HEP-PC.log)
  This is obviously NOT a major issue, but doessn't seem correct either.  Log attached.  Thanks!

w4sm, 2020-08-28:
Is there anything else I need to do?

NameOfTheDragon, 2021-01-13:
Closed due to inactivity (issue is over a year old)

closed 2021-01-13 by NameOfTheDragon