nexdome/Firmware #33: ArduinoSTL library not compatible with AVR boards v1.8.3

NameOfTheDragon, 2020-10-23:
See also:
- https://forum.arduino.cc/index.php?topic=696369.0
- https://github.com/mike-matera/ArduinoSTL/issues/56
- https://github.com/arduino/ArduinoCore-avr/issues/287

Projects targetting AVR boards that `#include <ArduinoSTL>` fail to compile when upgraded to boards version 1.8.3. See the linked discussions above for details.

Workaround: for the time being, NexDome firmware must be compiled with AVR Boards version 1.8.2. The problem should be resolved when boards package 1.8.4 is released.

NameOfTheDragon, 2021-01-13:
This no longer seems to be an issue.

closed 2021-01-13 by NameOfTheDragon

ghilios, 2021-10-06:
FWIW, this is still an issue. It depends on what order your frameworks were installed in. 1.8.4 still hasn't been released - I recommend adding this to the README.