nexdome/Firmware #38: Manual Rotator button no longer works with 4.0

ghilios, 2021-03-08:
After upgrading the firmware to 4.0.0, the manual rotation button on the physical device no longer works. ASCOM control still functions though.

NameOfTheDragon, 2021-03-09:
Thanks for reporting. I have had independent confirmation of the issue - it seems like it might be a timeout that may be related to the holding torque changes that went into the latest release. It's under investigation.

NameOfTheDragon, 2021-03-09:
Here is the offending code:

```lang=cpp
		// Release stepper holding torque if there has been no serial communication for "a long time".
		if (serialInactivityTimer.Expired())
			stepper.releaseMotor();
```

This runs in the main loop and once the timer has expired it will keep disabling the stepper motor every time around the loop. Any activity on the serial port resets the timer, which is why the ASCOM driver (et al) still work. It should be a simple fix.

ghilios, 2021-03-09:
Nice! Let me know if you'd like me to test it before you release.

NameOfTheDragon, 2021-03-09:
@ghilios @AstroFrank Please could you try the hex file attached to this pull request and let me know if it solves the problem?
https://github.com/nexdome/Firmware/pull/39/#issuecomment-794570409

NameOfTheDragon, 2021-03-09:
Yes please, I attached a hex file to the PR https://github.com/nexdome/Firmware/pull/39/#issuecomment-794570409
--T.

ghilios, 2021-03-11:
Sorry I was ill yesterday and couldn't get to it. I will try today after work!

NameOfTheDragon, 2021-03-11:
Thanks George, don't stress over it. My own tests suggest that the fix is good and I've had another independent confirmation. I'm going to issue a bugfix release later today.

--T.

closed 2021-03-12 by NameOfTheDragon