nexdome/Firmware #40: how to update to 4.0.1 firmware?

nevion, 2021-03-14:
The 4.0.0 ascom properties seems to only want to show 4.0.0 files and no actual paths - how do we update to 4.0.1?

NameOfTheDragon, 2021-04-02:
Download the two hex files from https://github.com/nexdome/Firmware/releases/latest and save them in `C:\NexDome-Firmware` along with the other hex files there. Then they will show up in the firmware update screen within the ASCOM driver.

nevion, 2021-04-07:
@NameOfTheDragon  got it, though I would add a file chooser based pathway too - it seems like the more common thing to do.

Also, while entirely unrelated, killing the huge clickable image in the control panel that takes you to nexdome, that thing needs to go.  I've clicked it so many times without wanting to go there or open a explorer window on my weak observatory controller PC...