nexdome/Firmware #41: Can't Connect after Update

w4sm, 2021-03-16:
I uploaded the 4.0.1 firmware to both my rotator and my shutter.  Both reported successful uploads.  However now I cannot connect to the dome.  I get the attached error message every time I try. The port# is correct.  I've tried cycling the power on everything with no success.  I also tried reinstalling 4.0.1 and also reverting to 4.0.0 with no success, although firmware is installed correctly according to the message.
I'm stumped.  The log just shows a failure to connect.  I have no problems installing the firmware.

w4sm, 2021-03-16:
Problem solved.  I needed to update to NexDomeControlSystem-4.0.0.exe as well as the firmware.  Everything worked fine after that.

NameOfTheDragon, 2021-04-02:
I'm glad you solved your problem. It's tempting to close the issue at that point, but this is probably worth keeping in open for a future version. The ASCOM driver checks for minimum compatible firmware version but it doesn't currently check for a maximum version. Based on the rules of semantic versioning, it should probably only accept one major version value or at least warn if the version is higher than expected.

I don't think this is high priority but I'm going to update the title of the issue and keep it open in case we make another version of the driver. Thanks for reporting this.